karawin/Ka-Radio #98: NodeMCU wifi not visible after flash

Brahypelma, 2022-03-19:
[image]
Hello . After flash nodeMcu with latest config I dont see any new wifi
Console log:
[image]
Checked with 2 different nodeMCUs. Pls help

karawin, 2022-04-03:
Check esp_init_default.bin

Brahypelma, 2022-04-08:
Oh I have NodeMCU with 4mb flash so need to pich in the ESP Downloader 32Mdit flash size. That resolved my issue

closed 2022-04-08 by Brahypelma

aravindnc, 2022-12-14:
@Brahypelma Can you explain what you did to fix the issue ?

Brahypelma, 2022-12-14:
@aravindnc I fixed my issue with setting correct flash size to flash it to right addresses. U need to pick up correct flash size

aravindnc, 2022-12-15:
@Brahypelma Thanks for the response. But the new flash download tool does not have the flash size selection feature. It seem to do it automatically.