karawin/Ka-Radio #106: Changing pins for VS1053  connection

h1aji, 2023-10-24:
Hi @karawin 
I would like to modify wires (to release pins 4, 5 and 15) as per below

> #define CS_PIN    0
> #define XDCS_PIN 16
> #define DREQ_PIN  9
> #define RST_PIN  10

and have modified vs1053.c
```
ICACHE_FLASH_ATTR void VS1053_HW_init() {
        printf(PSTR("\nVS1053 Init") );

        PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_DATA3_U, 3); //GPIO10
        PIN_FUNC_SELECT(PERIPHS_IO_MUX_SD_DATA2_U, 3); //GPIO9

        PIN_DIR_OUTPUT |= (1<<RST_PIN)|(1<<CS_PIN);
        gpio16_output_conf();
        PIN_DIR_INPUT |= (1<<DREQ_PIN);
        PIN_OUT_SET |= (1<<RST_PIN)|(1<<CS_PIN);
        gpio16_output_set(1);

        spi_init(HSPI);
        spi_clock(HSPI, 4, 10); //2MHz
        PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, 0); //GPIO0 is set as CS pin (Chip Select / Slave Select)
}

ICACHE_FLASH_ATTR void SDI_ChipSelect(uint8_t State){
        if(State) gpio16_output_set(0);
        else gpio16_output_set(1);
}
```
All compiles ok but there is no sound coming.
I tried it with this [Arduino module](https://github.com/baldram/ESP_VS1053_Library) and it works fine.

Would you be able to point what is wrong?

karawin, 2023-10-25:
The pin 16 is used for the external ram. May be a conflict.
You must see the log on the uart for indications and error messages.

h1aji, 2023-10-25:
thats not a problem, I set SPIRAM CS pin to 15
```
void extramInit() {
	char test[17]   = {"FFFFFFFF"};
	char testram[17]= {"01234567"};

	gpio15_output_conf();
	spi_clock(HSPI, 4, 10); //2MHz

        PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, 3); //GPIO15 is HSPI CS pin (Chip Select / Slave Select)

	bool externram = false;
	extramWrite(strlen(test), 0, testram);
	extramRead(strlen(test), 0, test);
        if (memcmp(test,testram,16) == 0) externram = true;       
        //printf(PSTR("\n=> extraram state: %d 0x%x %s\n"),externram,test[0],test );
        if (externram) {
		printf(PSTR("\nExternal RAM detected%c"),0x0d);
	} else {
		printf(PSTR("\nExternal RAM not detected%c"),0x0d);
	}
}
```